newaetech/chipwhisperer #423: Chip not connecting on Mac

rdaw6, 2022-10-31:
I'm trying to use a ChipWhisperer NANO on macOS Big Sur 11.6.8, and am getting the following issue:

[image]

The chip is connected via a thunderbolt 3 port with a USB C to USB adapter, and the USB light on the chip is lit up, so I assume this verifies it is connected.

I know there is a little less documentation on using chipwhisperer with Mac, so I tried using windows through bootcamp, but still received the same error.

alex-dewar, 2022-10-31:
Hi,

Does the green USB LED blink, or does it stay lit? If it's the latter, it might be related to https://github.com/newaetech/chipwhisperer/issues/237.

If the LED does blink, can you confirm that you have libusb  installed, then run `system_profiler SPUSBDataType` in a terminal?

alex-dewar, 2023-09-19:
Hi,

Any update on this?

alex-dewar, 2023-09-27:
Closing, as there's been no update for a while.

closed 2023-09-27 by alex-dewar

unixb0y, 2023-10-28:
I have the same issue with ChipWhisperer Pro (CW1200) and no matter which USB-C to USB-A adapter I use.
I think it's a software issue, since the CW shows up in System Information and in `system_profiler` without issues.