newaetech/chipwhisperer #426: `as_int` argument when capturing traces on ChipWhisperer-Lite gives wrong length

Stian108, 2022-11-29:
Using `as_int=True` on ChipWhisper-Lite gives a trace longer than `scop.adc.samples`. In my case a trace of length 5190 instead of 5000. It also seems like the traces are misaligned at the start. This seems to stem from the float data being aligned to the trigger in `processData` in `_OpenADCInterface.py`, but the int data is not aligned in the same way. This seems like unintended behavior to me, as I expected the only difference between `as_int=True` and `as_int=False` to be the normalization and type of data, as the documentation indicates. 

Thanks in advance for taking a look at this :)

alex-dewar, 2022-11-29:
Thanks for the report on this. Yeah this is definitely not expected behaviour. I'll post an update when I've fixed or had more time to look into this.

alex-dewar, 2022-12-11:
Quick update on this: I've fixed the issue. There was both a misalignment from not using the trigger part of the adc data and additional data that gets cut off during `scope.sc.readData()`.

I'm on a branch for the next ChipWhisperer release and will close this once I merge back into develop

closed 2023-01-11 by alex-dewar