newaetech/chipwhisperer #430: OverflowError while trying to setup chupwhisperer lite xmega

retro-c24, 2023-01-12:
import chipwhisperer as cw
scope = cw.scope()

---------------------------------------------------------------------------
OverflowError                             Traceback (most recent call last)
C:\Users\SOURAP\AppData\Local\Temp\ipykernel_17636\1654912759.py in <module>
      1 import chipwhisperer as cw
----> 2 scope = cw.scope()

d:\chipwhisperer5_64\cw\home\portable\chipwhisperer\software\chipwhisperer\__init__.py in scope(scope_type, name, sn, idProduct, bitstream, force, prog_speed, **kwargs)
    335     rtn : scopes.ScopeTypes = scope_type()
    336     try:
--> 337         rtn.con(**kwargs)
    338     except IOError:
    339         scope_logger.error("ChipWhisperer error state detected. Resetting and retrying connection...")

d:\chipwhisperer5_64\cw\home\portable\chipwhisperer\software\chipwhisperer\capture\scopes\OpenADC.py in con(self, sn, idProduct, bitstream, force, prog_speed, **kwargs)
    516         if cwtype == "cwhusky":
    517             self.sc._is_husky = True
--> 518         self.sc._setReset(True)
    519         self.sc._setReset(False)
    520 

d:\chipwhisperer5_64\cw\home\portable\chipwhisperer\software\chipwhisperer\capture\scopes\_OpenADCInterface.py in _setReset(self, value)
    361                 #Hack to adjust the hwMaxSamples since the number should be smaller than what is being returned
    362                 self.hwMaxSamples = self.numSamples() - 45
--> 363             self.setNumSamples(self.hwMaxSamples)
    364         else:
    365             self.setSettings(self.settings() & ~SETTINGS_RESET)

d:\chipwhisperer5_64\cw\home\portable\chipwhisperer\software\chipwhisperer\capture\scopes\_OpenADCInterface.py in setNumSamples(self, samples)
    380 
    381     def setNumSamples(self, samples):
--> 382         self.sendMessage(CODE_WRITE, ADDR_SAMPLES, list(int.to_bytes(samples, length=4, byteorder='little')))
    383         self.updateStreamBuffer(samples)
    384 

OverflowError: can't convert negative int to unsigned

alex-dewar, 2023-01-12:
Hi,

Do you get any other error messages? `self.numSamples()` is probably returning something that doesn't make sense (like 0), which might be the result of the FPGA not being programmed correctly.

Also, what ChipWhisperer version/commit and firmware version are you on?

Is this something that happens consistently and has your ChipWhisperer worked before?

Alex

retro-c24, 2023-01-23:
Umm i m using chipwhisperer lite with xmega target board and using the latest commit 5.7.0 ,
This board was bought like an year before and i haven't seen it work i don't think anyone even flashed it till now.
and there are no other errors

alex-dewar, 2023-01-23:
Can you try running the following before connecting and posting the resulting output?

```python
cw.naeusb_logger.setLevel(cw.logging.DEBUG)
cw.scope_logger.setLevel(cw.logging.DEBUG)
```

alex-dewar, 2023-08-23:
Hi, any update on this?

alex-dewar, 2023-09-19:
Closing, as there's been no update on this in a while.

closed 2023-09-19 by alex-dewar