newaetech/chipwhisperer #432: udev rules on systems without "plugdev" group

J08nY, 2023-02-15:
Hi all,
I am trying to use ChipWhisperer on an ArchLinux system that does not use the "plugdev" group (which seems to be a Ubuntu thing that spread to other distros). Hence, the udev rules distributed with ChipWhisperer do not make much sense as commit 1b0533d changed them to use that group, while before they were using the `"uaccess"` TAG.

I do not really understand udev rules so I am not sure what the solution is here, but I guess if the original udev rules work on systems without that group, some comment on this could be added into the udev file and maybe into some additional documentation? I am happy to contribute those changes if someone acks the approach. I will try to see whether these rules actually work on my distro in the meantime.

colinoflynn, 2023-02-16:
Hi @J08nY - thanks for flagging this! The `udev` stuff is always so close to perfect. It sounds like maybe we need two options, I'm guessing the rule change reflected some issues we/elsewhere was seeing. Will check with @alex-dewar who committed that change.

alex-dewar, 2023-02-16:
I remember `"uaccess"` being inconsistent with applying the correct permissions, especially when plugging in devices after logging in. Sometimes, I had to logout/login after plugging in a ChipWhisperer to get it to work, which is why I ended up just switching back.

I think the only distro-specific thing about this is that plugdev is a group that exists by default on some distros and not on others. If you create the group and join it, or change the group in the udev rules to one that you're already in, it should work the same. Perhaps it makes more sense to just use a "chipwhisperer" group instead and add the group creation to the install instructions.

colinoflynn, 2023-02-16:
If you have "both" does it break anything (e.g., have a file for ubuntu & a generic one)?

J08nY, 2023-02-16:
No clue what happens when both are used. Just two cents from the ArchLinux wiki: https://wiki.archlinux.org/title/Udev#Allowing_regular_users_to_use_devices

After creating the issue I noticed that the CW installation instructions for Linux do mention that the distro might not have the groups: https://chipwhisperer.readthedocs.io/en/latest/linux-install.html#installing-chipwhisperer and give advice that @alex-dewar gave here. So I guess a documentation update as I suggested is not necessary.

alex-dewar, 2023-02-17:
IMO it'd be better to either debug the issues with uaccess and just use that, or use the group based permissions, as having devices not work some of the time after plugging them in is pretty bad (or not at all as with [this forum post](https://forum.newae.com/t/oserror-unable-to-communicate-with-found-chipwhisperer-check-that-another-process-isnt-connected-to-it-and-that-you-have-permission-to-communicate-with-it/2585)).

I like changing over to a ChipWhisperer or NewAE group the best, as I think users of that group being able to access NewAE devices makes sense, it aligns with how this was done in pre-systemd arch: https://wiki.archlinux.org/title/Users_and_groups#Pre-systemd_groups, and makes the install instructions the same whether or not plugdev is a default group.

alex-dewar, 2023-03-02:
Closing this, as I've updated the install instructions to use a chipwhisperer group instead of plugdev

closed 2023-03-02 by alex-dewar