newaetech/chipwhisperer #433: Small issue with building CW308_STM32F3 HAL with ARM-GCC 12.2.0

Wmyers559, 2023-02-21:
Heya,

I was just (re)starting some work involving the chipwhisperer, and I found that I wasn't able to build the `simpleserial_base` target firmware (develop, commit 276df1615d4538d8f301dba220bf9e5d327e3efd), as the build was failing in `hal/stm32f3/stm32f3_sysmem.c` because the compiler didn't understand `caddr_t` as a type.

For me, that is defined in `sys/types.h`, and adding an include for that in that file allows the firmware to successfully build (with a warning about the output ELF file having a load section with `rwx` permissions). Precise GCC version is
```
arm-none-eabi-gcc (Arch Repository) 12.2.0
```

I'm glad to propose a pull request with this change if you think that it would be useful to merge/wouldn't break anything else (and I would also go through the other STM32 HALs to make sure that the issue doesn't show up there as well).

alex-dewar, 2023-02-22:
Hi,

Thanks for flagging this. I've replaced `caddr_t` with `char *` in 3420466765283daab41f330b4f1965f397fbb9a9. Let me know if that works for you.

Wmyers559, 2023-02-22:
Looks good to me! Thanks for the quick fix!

closed 2023-02-22 by Wmyers559