newaetech/chipwhisperer #435: CW305 Tutorial

hgordon-git, 2023-03-17:
The constraints file is causing DRC failures over the usb_clk_buf: Is the constraints file incorrectly configured? Synthesis and Implementation all work

1 out of 45 logical ports have no user assigned specific location constraint (LOC). This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all pin locations. This design will fail to generate a bitstream unless all logical ports have a user specified site LOC constraint defined.  To allow bitstream creation with unspecified pin locations (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks UCIO-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run.  Problem ports: usb_clk_buf.

colinoflynn, 2023-03-17:
As a quick fix, you could try rolling back before the [most recent commit](https://github.com/newaetech/chipwhisperer/commit/5bf482da14786816d2f92ecf869223871a6387b3), as this added support for a new board/interface system.

jpcrypt, 2023-03-20:
Sorry about that! Indeed last week's commit introduced the issue; today's commit fixes it properly.