newaetech/chipwhisperer #438: Missing software ChipWhisperer Capture

melona-debug, 2023-04-23:
Hello, why is the ChipWhisperer Capture not in the latest version of the ChipWhisperer I downloaded? 
I hope someone can help answer this question, thank you very much!

alex-dewar, 2023-04-23:
Hi,

The GUI for ChipWhisperer capture was removed in software version 5.0. If you would still like to use it, the final version with it available was 4.0.1: https://github.com/newaetech/chipwhisperer/releases/tag/v4.0.1

closed 2023-04-23 by alex-dewar

colinoflynn, 2023-04-24:
I forget if we had an explainer somewhere on this - but as Alex mentioned it was removed, which was because ChipWhisperer 4 (and the GUI) were exclusively Python-2.  The GUI used many features of PySide and was incompatible with Python3 support. In practice the GUI was useful for short demos but quickly become very complex to modify for a lot of "real" work.

As we have a small team we decided it was better to drop it entirely rather than keep it half-supported or similar.

Many features of the GUI are available via Jupyter notebooks, for example you can setup a notebook to provide a real-time plot. While it requires more setup effort, it does mean you can more easily modify it to suit your exact requirements. In practice we found even in e.g. running trainings users would quickly get to the point of needing to change the "provided" GUI. It was much more difficult to modify the old GUI code compared to the new Jupyter notebook based system.

melona-debug, 2023-04-24:
I understand now, thank you very much for your response!