newaetech/chipwhisperer #456: SAM4S Target Trace Divergence

alex-dewar, 2023-08-21:
The SAM4S seems to sometimes generate traces that look very different for a small portion of the AES algorithm. It's unclear if this is caused by the data that's being processed or some other reason. This doesn't seem to happen with the STM32F3 target with Husky.

[image]

alex-dewar, 2023-08-21:
Was caused by loop optimizations in the SAM4S. Disabling this fixes the issue. Will close when a fix is pushed

alex-dewar, 2023-08-22:
Fixed by 8b46fbd69867c1885066e4176e24f13ee6698db0

closed 2023-08-22 by alex-dewar

colinoflynn, 2023-08-22:
Nice catch! Had seen that for a while and never investigated, interesting there is still some flash/cache stuff hidden away a bit. For reference for anyone else, this is the SAM4S feature:
[image]