newaetech/chipwhisperer #469: Chipwhisperer installation and python dependencies

MKesenheimer, 2023-12-02:
To install the chipwhisperer software, I followed the instructions at [https://chipwhisperer.readthedocs.io/en/latest/linux-install.html](https://chipwhisperer.readthedocs.io/en/latest/linux-install.html).
However, when running the command
```
jupyter notebook
```
python errors occurred. I had to install the packages traitlets and param manually:
```
pip uninstall traitlets
pip install traitlets==5.9.0

pip uninstall param
pip install param==1.13.0
```

Please update the dependencies accordingly.

alex-dewar, 2023-12-04:
Thanks for the report on this. I'll get the dependencies updated.

It might be time to look at updating notebook to 7.0

alex-dewar, 2023-12-11:
I've updated the jupyter requirements to their latest version and things seem to work fine.