newaetech/chipwhisperer #470: AttributeError on Displaying Traces with ChipWhisperer-Lite on Ubuntu

abolfazlsajadi, 2023-12-05:
Hello ChipWhisperer Team,

I've encountered an issue with the ChipWhisperer-Lite on my Ubuntu system. I followed the installation guide closely using pyenv, as outlined in the ChipWhisperer Installation Guide for Ubuntu, and didn't run into any errors or warnings during the process.

**Issue Description:**
The problem arises when I attempt to display traces using the command `cw.plot(wave)`. Each time I do this, I encounter the following error:
`
AttributeError                            Traceback (most recent call last)
Cell In[9], line 1
**----> 1 cw.plot(wave)**

File ~/chipwhisperer/software/chipwhisperer/__init__.py:547, in plot(*args, **kwargs)
    545 if (len(args) == 0) and (len(kwargs) == 0):
    546     args = [[]]
**--> 547 import holoviews as hv # type: ignore**
    548 _default_opts = {'height': 600, 'width': 800, 'framewise': True, 'tools': ['hover'], 'active_tools': ['box_zoom']}
    549 hv.extension('bokeh', logo=False) #don't display logo, otherwise it pops up everytime this func is called.

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/holoviews/__init__.py:115
    111 warnings.filterwarnings("ignore",
    112                         message="elementwise comparison failed; returning scalar instead")
    114 if "IPython" in sys.modules:
**--> 115     from .ipython import notebook_extension**
    116     extension = notebook_extension # noqa (name remapping)
    117 else:

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/holoviews/ipython/__init__.py:16
     14 from ..element.comparison import ComparisonTestCase
     15 from ..util import extension
**---> 16 from ..plotting.renderer import Renderer**
     17 from .magics import load_magics
     18 from .display_hooks import display

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/holoviews/plotting/__init__.py:11
      9 from ..element import Area, Image, QuadMesh, Polygons, Raster
     10 from ..element.sankey import _layout_sankey, Sankey
**---> 11 from .plot import Plot**
     12 from .renderer import Renderer, HTML_TAGS # noqa (API import)
     13 from .util import apply_nodata, list_cmaps # noqa (API import)

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/holoviews/plotting/plot.py:17
     14 import numpy as np
     15 import param
**---> 17 from panel.config import config**
     18 from panel.io.document import unlocked
     19 from panel.io.notebook import push

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/panel/__init__.py:48
      1 """
      2 Panel is a high level app and dashboarding framework
      3 ====================================================
   (...)
     46 https://panel.holoviz.org/getting_started/index.html
     47 """
**---> 48 from . import layout  # noqa**
     49 from . import links  # noqa
     50 from . import pane  # noqa

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/panel/layout/__init__.py:31
      1 """
      2 Layout
      3 ======
   (...)
     29 https://panel.holoviz.org/getting_started/index.html
     30 """
**---> 31 from .accordion import Accordion  # noqa**
     32 from .base import (  # noqa
     33     Column, ListLike, ListPanel, Panel, Row, WidgetBox,
     34 )
     35 from .card import Card  # noqa

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/panel/layout/accordion.py:9
      5 import param
      7 from bokeh.models import Column as BkColumn, CustomJS
**----> 9 from .base import NamedListPanel**
     10 from .card import Card
     12 if TYPE_CHECKING:

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/panel/layout/base.py:17
     13 import param
     15 from bokeh.models import Column as BkColumn, Row as BkRow
**---> 17 from ..io.model import hold**
     18 from ..io.state import state
     19 from ..reactive import Reactive

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/panel/io/__init__.py:13
     11 from .logging import panel_logger  # noqa
     12 from .model import add_to_doc, diff, remove_root  # noqa
**---> 13 from .notebook import (  # noqa**
     14     _jupyter_server_extension_paths, block_comm, ipywidget, load_notebook,
     15     push, push_notebook,
     16 )
     17 from .profile import profile  # noqa
     18 from .resources import Resources  # noqa

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/panel/io/notebook.py:39
     37 from .embed import embed_state
     38 from .model import add_to_doc, diff
**---> 39 from .resources import (**
     40     PANEL_DIR, Bundle, Resources, _env, bundle_resources,
     41 )
     42 from .state import state
     44 if TYPE_CHECKING:

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/panel/io/resources.py:30
     27 from jinja2.loaders import FileSystemLoader
     28 from markupsafe import Markup
**---> 30 from ..config import config**
     31 from ..util import isurl, url_path
     32 from .state import state

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/panel/config.py:486
    483 else:
    484     _params = _config.param.params()
**--> 486 config = _config(**{k: None if p.allow_None else getattr(_config, k)**
    487                     for k, p in _params.items() if k != 'name'})
    490 class panel_extension(_pyviz_extension):
    491     """
    492     Initializes and configures Panel. You should always run `pn.extension`.
    493     This will
   (...)
    514     will be using the `FastListTemplate`.
    515     """

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/panel/config.py:261, in _config.__init__(self, **params)
    260 def __init__(self, **params):
**--> 261     super().__init__(**params)**
    262     self._validating = False
    263     for p in self.param:

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/param/parameterized.py:4139, in Parameterized.__init__(self, **params)
   4133 global object_count
   4135 # Setting a Parameter value in an __init__ block before calling
   4136 # Parameterized.__init__ (via super() generally) already sets the
   4137 # _InstancePrivate namespace over the _ClassPrivate namespace
   4138 # (see Parameter.__set__) so we shouldn't override it here.
**-> 4139 if not isinstance(self._param__private, _InstancePrivate):**
   4140     self._param__private = _InstancePrivate(
   4141         explicit_no_refs=type(self)._param__private.explicit_no_refs
   4142     )
   4144 # Skip generating a custom instance name when a class in the hierarchy
   4145 # has overriden the default of the `name` Parameter.

File ~/.pyenv/versions/3.9.5/envs/cw/lib/python3.9/site-packages/panel/config.py:348, in _config.__getattribute__(self, attr)
    341 """
    342 Ensures that configuration parameters that are defined per
    343 session are stored in a per-session dictionary. This is to
    344 ensure that even on first access mutable parameters do not
    345 end up being modified.
    346 """
    347 from .io.state import state
**--> 348 init = super().__getattribute__('initialized')**
    349 global_params = super().__getattribute__('_globals')
    350 if init and not attr.startswith('__'):

AttributeError: '_config' object has no attribute 'initialized'`

**Troubleshooting Steps Taken:**
- I've tried reinstalling the software on another system to rule out any system-specific issues, but the same error persists.

I am seeking guidance or suggestions on potential solutions to this problem. Any help would be greatly appreciated, as this issue is hindering my progress.

Thank you for your time and assistance.

Best regards,
Abish

alex-dewar, 2023-12-10:
Hi,

Can you try updating the jupyter requirements.txt packages and seeing if that fixes your issue:

```sh
cd jupyter
pip install -r requirements.txt --upgrade
```

abolfazlsajadi, 2023-12-12:
Unfortunately, updating the requirements.txt for Jupyter didn't resolve the issue. I encountered a ModuleNotFoundError related to 'jupyter_server.contents'. To fix this, I upgrade the notebook version to 6.5.6. While this adjustment resolved the module error, I'm still experiencing problems with cw.plot(wave) not functioning. However, I'm able to use plt for checking Waves.

alex-dewar, 2023-12-12:
Can you try grabbing the latest jupyter commit and running `pip install --upgrade --force-reinstall -r requirements.txt` instead?

abolfazlsajadi, 2023-12-13:
Thank you for your assistance

With your assistance, the problem has been resolved

closed 2023-12-13 by abolfazlsajadi