newaetech/chipwhisperer #472: Sources for release 5.7.0 identify as version 5.6.1?

twelho, 2023-12-12:
Hi :wave: 

Thanks for the great OSS hardware and project! While debugging, I was checking the `chipwhisperer` Python package version, which indicated version `5.6.1`, which is odd, since I'm installing from https://github.com/newaetech/chipwhisperer/archive/refs/tags/5.7.0.tar.gz on Linux. Since changes from 5.7.0, such as https://github.com/newaetech/chipwhisperer/commit/13f37a6406e7ef7f8a92ff1f62f0b4a7bc85ba74, are present (currently debugging a hang with `simpleserial-rsa`), was there simply a mistake in the packaging process?

[image]

Interestingly, the sources at https://pypi.org/project/chipwhisperer/5.7.0/#files do have the correct version in `setup.py`. Are all of the intended changes for 5.7.0 present in the sources of the 5.7.0 release?

alex-dewar, 2023-12-12:
Hi,

Yeah, the version probably just got updated after the tag was made, there were a few different locations for versions that have since been combined. Generally we recommend just grabbing the latest develop commit or using the installer if you're on Windows.

twelho, 2023-12-13:
Alright, I'm running on top of `devel` now, but the hang with `simpleserial-rsa` persists. Will debug a bit further, and then open a new issue for it.

closed 2023-12-13 by twelho