newaetech/chipwhisperer #486: Embedded assembly issues：my .S file been deleted

Miska-raligun, 2024-04-26:
Embedded assembly issues. I used some functions defined in assembly language (.S files) in my .c program, and placed them in the same directory as sampleserial-xxx , also adding them correctly to the makefile. However, when I compile using the acquisition script, these .S files are not compiled into .o files, and when I return to the original directory, I find that my .S files have been deleted. Why is this happening? How can I solve it? I am using the default makefile rules file Makefile.inc from ChipWhisperer.

alex-dewar, 2024-04-26:
Hi,

I'm not able to replicate this. Can you post your makefile?

Miska-raligun, 2024-04-26:
`# Target file name (without extension).
# This is the base name of the compiled .hex file.
TARGET = simpleserial-aesasm

# List C source files here.
# Header files (.h) are automatically pulled in.
SRC += simpleserial-aesasm.c
SRC += $(wildcard *.c) $(wildcard *.S)

# Use simpleserial 1
# SS_VER = SS_VER_1_1

# No crypto required
CRYPTO_TARGET = NONE

EXTRA_OPTS = NO_EXTRA_OPTS
CFLAGS += -D$(EXTRA_OPTS)
LDFLAGS += -specs=nosys.specs

#Add simpleserial project to build
include ../simpleserial/Makefile.simpleserial

FIRMWAREPATH = ../.
include $(FIRMWAREPATH)/Makefile.inc`
OK, most of it is templated; I only changed the file names. To compile for the STM32F1 platform, I added "LDFLAGS += -specs=nosys.specs". When I compile using a Makefile without this statement for the STM32F4 platform, I encounter the same issues mentioned above.

colinoflynn, 2024-04-26:
Hello,

This line in `makefile.inc` will remove the generated *.s files: [Makefile.inc Line 552](https://github.com/newaetech/chipwhisperer/blob/develop/hardware/victims/firmware/Makefile.inc#L552).

This should use the case-sensitive `rm` command (defined [here](https://github.com/newaetech/chipwhisperer/blob/develop/hardware/victims/firmware/Makefile.inc#L294)). But if for some reason it's using a non-case-sensitive (windows del) you'd see that.

You could try commenting out the Line 552 of Makefile.inc and see if that fixes it.

I suspect if not that there is somewhere it's missing the case sensitivity. Unfortunately that is a bit of a "known issue" on Windows sometimes (I remember this from avr-gcc/WinAVR a long time ago, but forget the specifics now).

You could also try putting your `.S` file in a subdirectory to see if that solves it.

Miska-raligun, 2024-05-07:
Thank you for the response! 
I finally resolved the issue by modifying makefile. By adding the path of the assembly files to ASRC+=$(wildcard *.S), they can now be compiled successfully and won't be deleted. I noticed this section in the makefile.inc file:
```
# List Assembler source files here.
# Always end them with a capital .S. Files ending in a lowercase .s
# will not be considered source files but generated files (assembler
# output from the compiler) and will be deleted upon "make clean"!
# Even though the DOS/Win* filesystem matches both .s and .S the same,
# it will preserve the spelling of the filenames, and gcc itself does
# care about how the name is spelled on its command-line.
ASRC +=
```