newaetech/chipwhisperer #493: fpga_write limited to 47 bytes

alphanumericnonsense, 2024-06-24:
In trying to write to a `cw305_aes_reg.v` style register with CW-Lite and CW305, I'm unable to write more than 47 bytes using the `target.fpga_write` function below.

In `CW305.py`
```python
    def fpga_write(self, addr, data):
        """Write to an address on the FPGA

        Args:
            addr (int): Address to write to
            data (list): Data to write to addr

        """
        if len(data) <= 0:
            raise ValueError("Invalid data {}".format(data))
        addr = addr << self.bytecount_size
        if self.platform in ['cw305', 'cw310', 'cw340']:
            return self._naeusb.cmdWriteMem(addr, data)
        elif 'ss2' in self.platform:
            payload = list(int.to_bytes(addr, length=4, byteorder='little'))
            payload.extend(data)
            self.ss2.send_cmd(cmd=0x23, scmd=0x57, data=payload)
            self._ss2_check_status()
```
which uses the following from `naeusb.py`
```python
    def cmdWriteMem(self, addr : int, data):
        """
        Send command to write memory over external memory interface to FPGA. Automatically
        decides to use control-transfer or bulk-endpoint transfer based on data length.
        """
        pload = util.get_bytes_memview(data)
        if len(pload) < NAEUSB_CTRL_IO_THRESHOLD:
            self._cmd_writemem_ctrl(addr, pload)
        else:
            self._cmd_writemem_bulk(addr, pload)

        naeusb_logger.debug("FPGA_WRITE: bulk: {}, addr: {:08X}, dlen: {:08X}, response: {}"\
            .format("yes" if len(pload) >= NAEUSB_CTRL_IO_THRESHOLD else "no", addr, len(pload), data))

        return None
```
and
```python
NAEUSB_CTRL_IO_THRESHOLD = 48
```
So something with `_cmd_writemem_bulk` isn't working I guess, I don't know.  I tried increasing the limit from 48 to 65 to no avail.

alex-dewar, 2024-06-26:
Hi,

What firmware is your CW305 on? This was definitely an issue on older firmware versions, but I had thought that I'd fixed that in an earlier firmware.

alex-dewar, 2024-10-29:
Hi, closing this as it hasn't had an update in a while. If you're still running into this, let me know.

closed 2024-10-29 by alex-dewar