newaetech/chipwhisperer #496: chipwhisperer-lite programming problems

HoulaiZhang, 2024-07-11:
By modifying the firmware code and programming, can the original data acquisition and signal processing functions of the FPGA be changed to AES encryption and data (plaintext, key, etc.) can be exchanged with the PC through the serial port?
If possible, which part of the file should I modify and compile to a .bin file and program to cwlite?

jpcrypt, 2024-07-15:
On CW-lite, the PC does not communicate directly with the FPGA; all communication is handled through the SAM3U processor. So, it's the SAM3U firmware that would need to change; not the FPGA code.  If you look at the [schematic](https://rtfm.newae.com/Capture/ChipWhisperer-Lite/#schematic), you'll see that there is a set of spare Rx/Tx pins that are broken out to the JP1 header. So you could, in theory, re-write the SAM3U firmware to move communication to this port. You would also need to make updates on the Python side. That is no small amount of work.

But if what you want is the "look and feel" of talking directly to the target's serial port from your PC, that exists today without any changes required!
https://www.newae.com/post/finally-real-serial-ports-with-the-latest-cw-software-update

Let me know if I've misunderstood your question.

HoulaiZhang, 2024-07-16:
> On CW-lite, the PC does not communicate directly with the FPGA; all communication is handled through the SAM3U processor. So, it's the SAM3U firmware that would need to change; not the FPGA code. If [...]

Thank you, you understand my question correctly! With your help and the discussion of our group, we have revised the idea of using CW.

closed 2024-07-16 by jpcrypt