newaetech/chipwhisperer #504: How do I change the clock frequency at which the target board runs the algorithm?

HoulaiZhang, 2024-11-25:
For example, using Husky to connect ICE40U, I set FREQ to a different value (e.g. 5) when generating the bin file. Then burn it to the target board to run, I found that when FREQ=20 and FREQ=5, the time consumed is the same, about 15us. This seems to indicate that the frequency changes have not taken effect.

colinoflynn, 2024-11-25:
If you are using the sample code, most of the targets use the external clock directly, so you don't need to change the frequency in the binary. Instead you can change the external clock with `scope.clock.clkgen_freq = 20E6` for example. You'll need to scale the target baud rate as well, set simpleserial to use version 1.1 (which has a default 38400 baud) & you could scale it like this:
```
print(target.baud)
target.baud = 20E6 / 7.37E6 * 38400
```

This just scales the baud rate by the same amount as you changed the target frequency. You may eventually need to change settings in the target binary as sometimes they need to know the external clock frequency for PLL settings (not sure on the iCE40, this may be the case as well). But you can at least adjust it some % of the normal without issue.

jpcrypt, 2024-11-25:
If you're referring to the `FREQ=20` in the [AES makefile](https://github.com/newaetech/chipwhisperer/blob/develop/firmware/fpgas/aes/icestorm/makefile#L3), this is the frequency that the FPGA toolchain is told to meet timing for, as it figures out how to translate Verilog to FPGA elements. It is independent of the clock frequency that the target is run at, which as Colin said, is specified via `scope.clock.clkgen_freq`.

HoulaiZhang, 2024-12-04:
Thank you！But I can't set clk less than 5MHz. In another Issues I learned that if we are not allowed to use the target clock as the source clock for these features, then we should be able to run Husky below 5 MHz.How do I set this up?

jpcrypt, 2024-12-04:
5 MHz is the minimum target clock frequency supported by Husky at the moment. 
We hope to be able to address this (https://github.com/newaetech/chipwhisperer/issues/502), but at this moment it's not possible.