newaetech/chipwhisperer #508: AES pipilened for Chipwhisperer CW305 - 35T

alimousios98x, 2025-01-15:
I am trying to integrate a custom extra logic into the dcp of the pipelined AES I mentioned but i need the name of the clock that is being used as a reference in order to fully route it. I see there are 3 clocks via the timing report summary of Vivado: pll_clk1, tio_clkin and usb_clk. Could u explain me their role and which one should I use.

[image]

jpcrypt, 2025-01-15:
Well, it depends what you want to do. `usb_clk` is used by the register interface. The AES core is clocked by either `pll_clk1` or `tio_clkin`. The former is the clock from the on-board PLL (which is what's used in our demo notebooks), the latter is the HS2 pin of the 20-pin connector (that could e.g. be supplied by a CW-lite). Which of those two clocks is used is controlled by the CW305's switches and a register. It's quite straightforward if you follow the code.

You'll find this document useful: https://media.newae.com/appnotes/NAE0010_Whitepaper_CW305_AES_SCA_Attack.pdf

closed 2025-01-15 by jpcrypt

alimousios98x, 2025-01-16:
I need to get the name of the clock that is being driven by the BUFG as my code expects the clock net to be at the top level of the top cell. 
This is the error i receive:
[image]

This is the part of my code that needs a reference of the clock.
[image]