newaetech/chipwhisperer #509: Connecting my extra circuit added to the AES PIPELIEND DCP with a clock

alimousios98x, 2025-01-17:
As you mentioned in my previous issue there are indeed three clocks but it seems that they are limited inside of a site as their ROUTE_STATUS is INTRASITE and I cant connect them to the extra circuit. For example pll_clk1:

[image]

I explored the netlist schematic and found past the IBUF and the BUFGCTRL, a clock net that is being used by other parts of the circuit:

[image]

So is the **crypt_clk** an appropriate input clock for my extra circuit since this one seems to be used by other parts of the design as well? I dont see any other that i am able to route to my circuit.

alimousios98x, 2025-01-20:
Also when i do timing reports the added circuit that is connected to crypt_clk isn't being shown to any path. Timing report shows information regarding usb_clk, tio_clkin and async_default for a pushbutton but nothing for crypt_clk.

alimousios98x, 2025-01-20:
Basically i want to integrate a clock glitch detection circuit so I believe I want to route it with the internal PLL Clock but I cant do this directly. I am attempting to do it with crypt_clk (but this seems to be a multiplexer that takes other clocks too) but it doesn't seem to show any results at timing reports.

jpcrypt, 2025-01-20:
Simple exploration of the design shows that there are two input clocks (pll_clk1 and tio_clkin) which go to a selection mux, the output of which is crypt_clk, which is the clock for the AES logic.
I don't know how to help you further  here: I am happy to answer questions about the CW305 and our example FPGA target designs, which I've already done, but I cannot help you integrate some other design. Our example FPGA target designs are provided as-is, and I'll help you understand them, but the rest of the work falls on you.
None of the issues that you've filed on this subject are ChipWhisperer issues; the appropriate place for these types of questions would be general Xilinx or FPGA support forums. You can also post questions on our [forum](forum.newae.com) or our [discord](https://discord.gg/chipwhisperer).

closed 2025-01-20 by jpcrypt