newaetech/chipwhisperer #510: Unable to connect Chipwhisperer HuskyPlus

thisisish, 2025-01-22:
On connecting the device and running the script I am getting following error. Am I missing something?

`import chipwhisperer as cw
scope = cw.scope()`

`---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
Cell In[2], line 2
      1 import chipwhisperer as cw
----> 2 scope = cw.scope()

File c:\users\shout\chipwhisperer5_64\cw\home\portable\chipwhisperer\software\chipwhisperer\__init__.py:397, in scope(scope_type, name, sn, idProduct, bitstream, force, prog_speed, **kwargs)
    394         raise ValueError
    396 if scope_type is None:
--> 397     scope_type = get_cw_type(**kwargs)
    398 rtn : scopes.ScopeTypes = scope_type()
    399 try:

File c:\users\shout\chipwhisperer5_64\cw\home\portable\chipwhisperer\software\chipwhisperer\common\utils\util.py:578, in get_cw_type(sn, idProduct, hw_location, **kwargs)
    575     possible_ids = [0xace0, 0xace2, 0xace3, 0xace5, 0xace6]
    577 cwusb = NAEUSB_Backend()
--> 578 device = cwusb.find(serial_number=sn, idProduct=possible_ids, hw_location=hw_location)
    579 name = device.getProduct()
    580 cwusb.usb_ctx.close()

File c:\users\shout\chipwhisperer5_64\cw\home\portable\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py:413, in NAEUSB_Backend.find(self, serial_number, idProduct, hw_location)
    410 def find(self, serial_number : Optional[str]=None, idProduct : Optional[List[int]]=None, 
    411     hw_location : Optional[Tuple[int, int]]=None) -> usb1.USBDevice:
    412     # check if we got anything
--> 413     dev_list = self.get_possible_devices(idProduct, attempt_access=(not hw_location))
    414     if len(dev_list) == 0:
    415         raise OSError("Could not find ChipWhisperer. Is it connected?")

File c:\users\shout\chipwhisperer5_64\cw\home\portable\chipwhisperer\software\chipwhisperer\hardware\naeusb\naeusb.py:536, in NAEUSB_Backend.get_possible_devices(self, idProduct, dictonly, attempt_access)
    533                 dev_list.remove(dev)
    535         if len(dev_list) == 0:
--> 536             raise OSError("Unable to communicate with found ChipWhisperer. Check that \
    537 \nanother process isn't connected to it and that you have permission to communicate with it.")
    539         return dev_list

OSError: Unable to communicate with found ChipWhisperer. Check that 
another process isn't connected to it and that you have permission to communicate with it.
`

thisisish, 2025-01-22:
[image]

alex-dewar, 2025-01-22:
Hi,

This error means that the device is there, but there's some sort of OS level communication issue with the device. This mostly happens when you try to connect while you're still connected on another notebook. Have you been able to successfully connect before? I recommend unplug + replugging before running the above line.

closed 2025-01-23 by thisisish