newaetech/chipwhisperer #511: enabling RDP on Chipwhisperer Nano target STM32F0

gcarmix, 2025-01-22:
Hi, I would like to test the STM32F0 target available on the Chipwhisperer Nano for the RDP feature.
So I would like to enable readout protection at level 1 on the target. 
Is there a command from the chipwhisperer python API to do such operation?

Thank you,

Carmix

alex-dewar, 2025-01-22:
No, this functionality isn't included in our bootloader communication code, you'll have to consult with ST's reference material on enabling this. Probably the easiest way would be to write some C code to write to the appropriate flash address. You could also try using the lower level write functions in `hardware/naeusb/programmer_stm32fserial.py` (should be accessible via `scope.serialstm32f`).

closed 2025-01-22 by alex-dewar

gcarmix, 2025-01-24:
Thank you for your answer alex, in fact I tried to use the low level write functions and they seem to work correctly, 
just sending:
scope.serialstm32f.cmdReadoutProtect() does the trick and the chip does not allow me to read its memory using the command scope.serialstm32f.readMemory(0x8000000,4). 
Now I would like to test the chip memory RDP for vulnerability and see if when I apply power to the chip is it possible to read at least one memory word. I'm referring to the CVE-2017-18347 vulnerability: https://vuldb.com/?id.123931.
Do you have any hint on how to replicate this with CWNANO, do you think is it possible?