newaetech/chipwhisperer #515: Issue with cw.target() Programming CW305 Bitstream

my-rooftop, 2025-02-18:
Hello,

I’m trying to program my own bitstream onto the CW305 board, but it doesn’t seem to work.

After investigating the issue, I found that as **@jpthibault** suggested, I should try:

`target = cw.target(None, cw.targets.CW305, bsfile="/path/to/bitstream/", force=True)`

So I followed this approach and set:

`target = cw.target(None, cw.targets.CW305, bsfile="/home/boochoo/hqc/dummy-insertion-sparse-polymult/CW305/jupyter/cw305_top.bit", force=True)
`
However, I noticed an issue in **CW305.py** at **line 513**, which contains the following:

`status = self.fpga.FPGAProgram(bsfile, exceptOnDoneFailure=False, prog_speed=prog_speed)`

The problem is that FPGAProgram() expects a **file object**, but bsfile is being passed as a **string path** instead. This causes the function to fail.

To temporarily fix this, I modified **CW305.py:513** to:

`status = self.fpga.FPGAProgram(open(bsfile, "rb"), exceptOnDoneFailure=False, prog_speed=prog_speed)`

With this change, programming the bitstream works correctly.

Could you please review this issue and confirm if this is the intended behavior? It seems like bsfile should be properly handled before passing it to FPGAProgram().

jpcrypt, 2025-02-18:
Thank you for noting this- this was inadvertently broken by a recent commit, fixed now.

closed 2025-02-18 by jpcrypt

my-rooftop, 2025-02-18:
Thank u !