newaetech/chipwhisperer #520: Scope latest firmware API is broken

antoniovazquezblanco, 2025-03-17:
```Python
>>> import chipwhisperer as cw
>>> scope = cw.scope()
>>> scope.fw_version_str
'0.65.0'
>>> scope.latest_fw_str
'0...0'
>>> scope.latest_fw       
{'major': '0', 'minor': '.'}
>>> scope._getFWPy() 
'0.65.0'
>>>
```

I think this breaks https://github.com/newaetech/chipwhisperer-jupyter/blob/main/ChipWhisperer%20Firmware%20Upgrade.ipynb...

alex-dewar, 2025-03-17:
Good catch! I'll get that fixed up, probably caused by the replacement of the old .py firmware files

antoniovazquezblanco, 2025-03-18:
I don't think so...

Checkout this (redacted paths...):
```
PS C:\...\chipwhisperer-jupyter> .\.venv\Scripts\Activate.ps1
(.venv) PS C:\...\chipwhisperer-jupyter> pip install --upgrade chipwhisperer          
Requirement already satisfied: chipwhisperer in c:\...\chipwhisperer-jupyter\.venv\lib\site-packages (5.7.0)
Requirement already satisfied: configobj in c:\...\chipwhisperer-jupyter\.venv\lib\site-packages (from chipwhisperer) (5.0.9)
Requirement already satisfied: pyserial in c:\...\chipwhisperer-jupyter\.venv\lib\site-packages (from chipwhisperer) (3.5)
Requirement already satisfied: numpy<=1.26.4 in c:\...\chipwhisperer-jupyter\.venv\lib\site-packages (from chipwhisperer) (1.26.4)
Requirement already satisfied: libusb1 in c:\...\chipwhisperer-jupyter\.venv\lib\site-packages (from chipwhisperer) (3.3.0)
Requirement already satisfied: ECpy in c:\...\chipwhisperer-jupyter\.venv\lib\site-packages (from chipwhisperer) (1.2.5)
Requirement already satisfied: fastdtw in c:\...\chipwhisperer-jupyter\.venv\lib\site-packages (from chipwhisperer) (0.3.4)
Requirement already satisfied: Cython in c:\...\chipwhisperer-jupyter\.venv\lib\site-packages (from chipwhisperer) (3.0.12)
Requirement already satisfied: tqdm in c:\...\chipwhisperer-jupyter\.venv\lib\site-packages (from chipwhisperer) (4.67.1)
Requirement already satisfied: colorama in c:\...\chipwhisperer-jupyter\.venv\lib\site-packages (from tqdm->chipwhisperer) (0.4.6)
(.venv) PS C:\...\chipwhisperer-jupyter> pip install --force-reinstall --upgrade chipwhisperer
Collecting chipwhisperer
  Using cached chipwhisperer-5.7.0-py3-none-any.whl.metadata (17 kB)
Collecting configobj (from chipwhisperer)
  Using cached configobj-5.0.9-py2.py3-none-any.whl.metadata (3.2 kB)
Collecting pyserial (from chipwhisperer)
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting numpy<=1.26.4 (from chipwhisperer)
  Using cached numpy-1.26.4-cp310-cp310-win_amd64.whl.metadata (61 kB)
Collecting libusb1 (from chipwhisperer)
  Using cached libusb1-3.3.0-py3-none-win_amd64.whl.metadata (16 kB)
Collecting ECpy (from chipwhisperer)
  Using cached ECPy-1.2.5-py3-none-any.whl.metadata (3.5 kB)
Collecting fastdtw (from chipwhisperer)
  Using cached fastdtw-0.3.4-py3-none-any.whl
Collecting Cython (from chipwhisperer)
  Using cached Cython-3.0.12-cp310-cp310-win_amd64.whl.metadata (3.6 kB)
Collecting tqdm (from chipwhisperer)
  Using cached tqdm-4.67.1-py3-none-any.whl.metadata (57 kB)
Collecting colorama (from tqdm->chipwhisperer)
  Using cached colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
Using cached chipwhisperer-5.7.0-py3-none-any.whl (11.4 MB)
Using cached numpy-1.26.4-cp310-cp310-win_amd64.whl (15.8 MB)
Using cached configobj-5.0.9-py2.py3-none-any.whl (35 kB)
Using cached Cython-3.0.12-cp310-cp310-win_amd64.whl (2.8 MB)
Using cached ECPy-1.2.5-py3-none-any.whl (43 kB)
Using cached libusb1-3.3.0-py3-none-win_amd64.whl (140 kB)
Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Using cached tqdm-4.67.1-py3-none-any.whl (78 kB)
Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Installing collected packages: pyserial, libusb1, ECpy, numpy, Cython, configobj, colorama, tqdm, fastdtw, chipwhisperer
  Attempting uninstall: pyserial
    Found existing installation: pyserial 3.5
    Uninstalling pyserial-3.5:
      Successfully uninstalled pyserial-3.5
  Attempting uninstall: libusb1
    Found existing installation: libusb1 3.3.0
    Uninstalling libusb1-3.3.0:
      Successfully uninstalled libusb1-3.3.0
  Attempting uninstall: ECpy
    Found existing installation: ECPy 1.2.5
    Uninstalling ECPy-1.2.5:
      Successfully uninstalled ECPy-1.2.5
  Attempting uninstall: numpy
    Found existing installation: numpy 1.26.4
    Uninstalling numpy-1.26.4:
      Successfully uninstalled numpy-1.26.4
  Attempting uninstall: Cython
    Found existing installation: Cython 3.0.12
    Uninstalling Cython-3.0.12:
      Successfully uninstalled Cython-3.0.12
  Attempting uninstall: configobj
    Uninstalling configobj-5.0.9:
      Successfully uninstalled configobj-5.0.9
  Attempting uninstall: colorama
    Found existing installation: colorama 0.4.6
    Uninstalling colorama-0.4.6:
      Successfully uninstalled colorama-0.4.6
  Attempting uninstall: tqdm
    Found existing installation: tqdm 4.67.1
    Uninstalling tqdm-4.67.1:
      Successfully uninstalled tqdm-4.67.1
  Attempting uninstall: fastdtw
    Found existing installation: fastdtw 0.3.4
    Uninstalling fastdtw-0.3.4:
      Successfully uninstalled fastdtw-0.3.4
  Attempting uninstall: chipwhisperer
    Found existing installation: chipwhisperer 5.7.0
    Uninstalling chipwhisperer-5.7.0:
      Successfully uninstalled chipwhisperer-5.7.0
Successfully installed Cython-3.0.12 ECpy-1.2.5 chipwhisperer-5.7.0 colorama-0.4.6 configobj-5.0.9 fastdtw-0.3.4 libusb1-3.3.0 numpy-1.26.4 pyserial-3.5 tqdm-4.67.1
(.venv) PS C:\...\chipwhisperer-jupyter> python
Python 3.10.11 (tags/v3.10.11:7d4cc5a, Apr  5 2023, 00:38:17) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import chipwhisperer as cw
>>> scope = cw.scope()
>>> scope.latest_fw_str
'0...0'
>>> scope.latest_fw     
{'major': '0', 'minor': '.'}
>>> scope._getFWPy() 
'0.65.0'
>>> ^Z

```

antoniovazquezblanco, 2025-03-18:
I am running this on Windows 11, Python 3.10.11. The device is a CW Lite.

Any other way I may help?

alex-dewar, 2025-03-18:
Nevermind, this should be fixed in the latest commit.

antoniovazquezblanco, 2025-03-18:
Nice! Thanks for the fast response! :D

closed 2025-03-18 by antoniovazquezblanco